joypad-ai/joypad-os #6: 6 buttons mode broken ?

onthewaveagain, 2023-03-02:
The 6 buttons mode does not look to work (for me).

Tested on a real PC Engine with SFII'.

With a real Nec Avenue 6 : Toggling Mode A/B switch change the controller behaviour from 2 buttons to 6.
This can be observed in the options Menu. The buttons re-mapping list change automatically from displaying 2 buttons or 6.

With a 8bitdo PCE and an 8Bitdo M30 2.4, holding RUN and quickly tapping UP does nothing.

RobertDaleSmith, 2023-03-03:
I wrote the firmware so that only controllers with enough buttons should be capable of enabling it. So the 8BitDo PCE 2.4g is one where I expected it to not able to enable six button mode. But the 8BitDo M30 2.4g should work. I specifically bought one to add support in a recent firmware update.

Are you using the M30 with the wireless receiver or wired directly to the adapter?

I know the PCE2.4 controller has undocumented modes you can switch to. Maybe the M30 has similiar modes to toggle between XInput and Genesis/MegaDrive modes.

```
Undocumented PCE 2.4g mode options. Hold for 5 seconds (until led on receiver blinks).
Run+⬆️ = XInput Mode
Run+⬅️ = Switch Mode
Run+⬇️ = PCE Mini Mode
```
Possibly holding Start+Down will kick the receiver/controller itself into the Genesis/MegaDrive mode that I implemented with the correct 6 button mapping. If there is an XInput mode on the M30, then it will work but button mapping may not be 1:1 with NEC Avenue layout.

Another thing that could help troubleshoot is checking the VID/PID (vendor/product IDs) of the device when connected to a Mac/PC by checking the detected device details. For the M30, my firmware is expecting `vendor_id = 0x0ca3` and `product_id = 0x0024`.

onthewaveagain, 2023-03-03:
That is clear, think I know what is wrong.

I am pairing my M30 2.4 with the PCE dongle. The dongle does not work in 6-button mode when paired with a M30 (tested on a PC and think it is a shame). And regardless, like you said your firmware would not handle it as it recognises it as PCE dongle with 2-button only capable.

I did test the M30 2.4 in wired directly though and it does not look to recognise it at all. I tested on a PC and it works well, after checking vendor_id and product_id I think that is the issue. I have vendor_id = 045e and product_id = 028e.
What M30 did you use  ?

closed 2023-04-07 by RobertDaleSmith