joypad-ai/joypad-os #7: Support Krikzz Turbo Everdrive Pro's In-Game menu / hot-key combos

isermann, 2023-04-05:
Hello,

Several user's (including myself) have issues with accessing the In-Game menu on the new Turbo Everdrive Pro via the defined hot-key combos when using the USB-2-PCE adapter.
The combos simply don’t work with the adapter, neither for quick save/load nor bringing up the menu. There are reports for CoreGrafx 1 & 2 at least.
I tried with adapter’s March and December firmware w/o luck.

Krikzz just stated on Twitter: "I don not control controllers refresh timings, i just sniff what console reads. I don't think that i can do something with non regular controllers on my side."

Could you please look into this issue? Thanks!

RobertDaleSmith, 2023-04-05:
Thanks for bringing this to my attention. I don't have a Turbo Everdrive Pro on hand but I will see what I can figure out. The adapter should behave pretty close to 1:1 as original controllers so just need to narrow down the cause and I will patch it up.

I will need to do some research on Turbo Everdrive Pro to see what the hot key combos are. If the overlap with my own button combo for 6-button mode toggle, then that may be the issue. If so that will be an easy fix and I can just change the combo I use.

What are the button combos?

isermann, 2023-04-05:
Thanks for the very quick response!

You can find the manual [here](https://krikzz.com/pub/support/turbo-everdrive/pro-series/turbo-ed-pro-manual.pdf).

But in general, the default combos are: 

- Save: RUN+RIGHT
- Load: RUN+LEFT
- Menu: RUN+DOWN

RobertDaleSmith, 2023-04-05:
Cool, thanks for the details. Is it primarily just the Menu hot key or all three not working correctly?

isermann, 2023-04-05:
All three are not working.

Only thing working is RUN+SELECT to reset game, but don't know, if this is a feature/bug of PC Engine, though?

RobertDaleSmith, 2023-04-05:
I will dig in and see what I can find. One combo does overlap with my 2-button mode switching but I wouldnt expect it to cause this so for sure something is off. The Turbo Everdrive Pro is out of stock everywhere, but are you aware of this issue existing on the older Turbo Everdrive as well? Just trying to see what I can pick up to get this resolved the quickest.

isermann, 2023-04-05:
Thanks for you dedication!

The old Turbo Everdrive is pretty bare bones and has no in-game menu or any button combo.
[Here ](https://krikzz.com/our-products/cartridges/turbo-everdrive-pro.html) you can find a nice comparison table.

If I can support testing, please let me know!

isermann, 2023-04-06:
The original multitap + wired controller is also not working with TED Pro's in-game combos.
Maybe you can compile a test firmware where your adapter does not appear as a multitap?

RobertDaleSmith, 2023-04-06:
tldr: fix being written and tested now. will upload firmware update shortly.

---

@isermann this is a great find! I was actually speculating that this may be the reason (multitap related bug). This also explains the reports I have got that it works in some games and not others. Because the way the controllers is polled is determined by the game itself.

The USB2PCE behaves as if it is a multitap with a single controller attached. This was done for USB hub support as each additional controller outputs as the next player on the tap. 

Good news is this should be a relatively easy fix. When a game supports multitap reading but a multitap is not connected, then the single controller appears to the console as a multitap with the same controller connected to all 5 polls. So when USB2PCE detects a single USB controller, I can just make it behave the same way.

This should get around this bug, but for sure a bug still exist for multitaps and the Turbo Everdrive Pro itself.

RobertDaleSmith, 2023-04-06:
New released published here to try. If the multi-tap is the issue, then this will fix it:

https://github.com/RobertDaleSmith/usb2pce/releases/tag/USB2PCE_2023_04_06

isermann, 2023-04-06:
All default combos work fine now! Outstanding support from you!
I'll test some more tomorrow.

Now Krikzz might fix the multitap detection on his side?

RobertDaleSmith, 2023-04-06:
Yea he should be able to patch it up for multi-taps on his end.

After some testing I may have introduced new bugs. In SF2, it is now always seeing two players. So there must be something I'm missing to correctly identify that it is not a multi-tap to that game.

isermann, 2023-04-07:
Maybe you can add another of your hot-key combos to toggle between Turbo Everdrive Pro "mode" and previous behavior?

RobertDaleSmith, 2023-04-10:
Okay I have managed to put together a possible fix that doesn't cause issues with Street Fighter 2. @isermann would you mind testing to see if it still resolves the issues with Turbo EverDrive Pro?

[USB2PCE_2023_04_10.uf2.zip](https://github.com/RobertDaleSmith/usb2pce/files/11188780/USB2PCE_2023_04_10.uf2.zip)

isermann, 2023-04-10:
Great! I tried it and it works, also SF2 is working with hot-key combos + your adapter on TED Pro.

I also discovered, that in general - also with wired controller - the combos don't work in "Dragon Spirit".
In "The Legendary Axe" the hot-keys are swapped clockwise, e.g., RIGHT -> DOWN, DOWN -> LEFT.

Thanks again!

isermann, 2023-04-11:
Hot-keys **not** working in _Dragon Spirit_.

In _The Legendary Axe_ swapped keys are:
- RIGHT = Load-State (should be: Save-State)
- DOWN = Save-State (should be: Menu)
- LEFT = Menu (should be: Load-State)
- UP --> not used

RobertDaleSmith, 2023-04-11:
I took a deeper look at Dragon Spirit and it for sure polls the controllers in an unusual way compared to most games. But The Legendary Axe looks pretty standard on the oscilloscope for the controller IO lines. 

Krikzz mentioned on Twitter that he is doing some really tricky techniques to sniff the controller data through the CPU. If he is having to do is this way, then it must be incredibly difficult to detect the button combo from his end. Especially when the games used nonstandard controller polling patterns like Dragon Spirit. But then again The Legendary Axe looks standard on the output end, yet CPU readings appear to be rotating the button combo dpad directions. I suspect its related to combo of how the individual game is written and the EverDrive's algo for detecting buttons presses.

closed 2023-04-16 by RobertDaleSmith