joypad-ai/joypad-os #8: Firmware usb2pce-2023-06-14 and greater break compatibility with 8bitdo PCE controllers

mario-64, 2023-06-23:
Firmware usb2pce-2023-06-14 and usb2pce-2023-06-21 break compatibility with 8bitdo PCE controllers. Specifically, dpad Up and dpad Left no longer work. There may be other issues too but this is as much as I could test having no dpad left or up. Verified using two different 8bitdo PCE controllers. Reverted back to the June 8 firmware and it's working again.

RobertDaleSmith, 2023-06-24:
Thanks so much for bringing this to my attention. This is the last controller I wanted to break support for. LOL

Good news is that I traced the issue down and fixed it. Appears to be related to the introduction of Switch controller support. Some how the receiver goes into Switch mode now that I support it and the third-party switch mode had a bug related to the conversion from analog values to dpad directions.

The update with the fix will be live shortly. But a work-around in the meantime is making sure your controller itself is set to default dpad mode rather than simulated left analog. 

## PCE Default Dpad Mode:
Hold `SELECT + UP` on the controller for about five seconds. You should see the LED flash and then the dpad will now send normal dpad buttons.

Undocumented 8BitDo PCE 2.4g options -> https://twitter.com/RobertDaleSmith/status/1570563632114831360

mario-64, 2023-06-24:
OMG! So my controller is in some weird mode? Sheesh I had no idea. Don’t even know how it got that way.

RobertDaleSmith, 2023-06-24:
This issue should be resolved in the [latest](https://github.com/RobertDaleSmith/usb2pce/releases/tag/USB2PCE_2023_06_24) release.

closed 2023-06-24 by RobertDaleSmith