joypad-ai/joypad-os #12: 8bitdo M30 2.4G for Genesis Mini wireless 6-button mode issues

DeerSteak, 2023-07-29:
In Street Fighter II CE (or other 6-button enabled games) the buttons on the 8bitdo M30 2.4G for Genesis Mini have their buttons mapped incorrectly. My guess is that since this controller will work with the Switch, it's falling into Switch mode. 

X = III
Y = IV
Z = V
A = I
B = II
C = VI

In addition, down diagonals do not seem to register, and on the Options screen you can see the game flicker between 6-button and 2-button modes. 

All of this is true whether the controller is connected via the USB cable directly to the USB2PCE or connected via the 2.4 GHz wireless receiver. 2-button mode seems fine other than the diagonals. 

I am using the latest firmware for both the receiver and the controller from 8bitdo's site (3.12 for the receiver, 2.08 for the controller) and the July 4 firmware for the USB2PCE.

RobertDaleSmith, 2023-07-30:
For the M30 2.4 to have the correct button layout, it will need to be in d-input mode. If it’s in any other mode like Switch or Xinput then it’s not possible for the adapter to know it’s a six button layout over a normal Xbox or switch controller.

Changing the input mode is not well documented in the manual. But you should be able to change to d-input mode by holding the 8BitDo icon button and down for about ten seconds. Then the PCE 1 button will be mapped to C.

I will circle back this evening and double check the down diagonals.

RobertDaleSmith, 2024-01-12:
This should be resolved in latest firmwares.

closed 2024-01-12 by RobertDaleSmith