joypad-ai/joypad-os #15: Verify Xbox One Third-party controller compatibility

RobertDaleSmith, 2023-07-31:
Here is a possible list of VIDs of Xbox One controllers. Microsoft, PDP, and PowerA should be covered but I need to verify all third-party controllers are working.

- [ ] {(0x045e) }, /* Microsoft */
- [ ] {(0x0738) }, /* Mad Catz */
- [ ] {(0x0e6f) }, /* PDP */
- [ ] {(0x0f0d) }, /* Hori */
- [ ] {(0x1532) }, /* Razer */
- [ ] {(0x24c6) }, /* PowerA */
- [ ] {(0x20d6) }, /* BDA */
- [ ] {(0x044f) }, /* Thrustmaster */
- [ ] {(0x10f5) }, /* Turtle Beach */
- [ ] {(0x2e24) }, /* Hyperkin */
- [ ] {(0x3285) }, /* Nacon */
- [ ] {(0x2dc8) }, /* 8BitDo */
- [ ] {(0x2e95) }, /* SCUF */
- [ ] {(0x3537) }, /* GameSir */

RobertDaleSmith, 2026-02-19:
Should already work with the USB host input using the 2.4Ghz dongle

lucaslealdev, 2026-02-21:
More context with videos. When plugged the dongle blinks and stops like is doing some kind of handshake and failing (Maybe?) and no usb debug info shows up in the config window.

When plugging the controller directly using its USB cable it works tho. Very strange behavior.

[image]
[image]