joypad-ai/joypad-os #24: Black screen when playing Youkai Douchuuki

p0ojisan, 2024-01-17:
I'm new user of USB2PCE with v1.0.3 firmware. In most of titles, it works correctly, but found an issue when playing Youkai Douchuuki.

When I turn on PC Engine with USB2PCE plugged in, the screen remains black. In this situation, any controller input doesn't work at all. When I unplug USB2PCE while the power stays ON, the title screen of Youkai Dochuki is correctly displayed.

I posted video of this issue in X.
https://twitter.com/peca_port0/status/1747291292156277220

Seems like, USB2PCE repeats sending soft reset, so that the screen remains black. But not sure. You can also reproduce this by putting Youkai Douchuuki to Turbo EverDrive Pro, as I tried. 

Is there any solution to avoid this?

RobertDaleSmith, 2024-01-17:
Does this only occur with this one game for you? Or any game?

I will try this specific game on my end and see if I can reproduce the issue. If it’s game specific then we should be able to patch it some how.

RobertDaleSmith, 2024-01-17:
Okay I was able to do some testing and unfortunately I was unable to replicate the issue. I tested Youkai Douchuuki on my PCEngine with a Super SD System 3, and the game boot and played without issue.

I'm very curious to find out more about your setup though to see if we can find a solution together. I wonder if this is a specifc issue when using USB2PCE + Turbo EverDrive Pro + Youkai Douchuuki. I'm guessing you tested the physical copy of the game without the EverDrive?

Also you mentioned the console does not boot when USB2PCE is connected.. Does the EverDrive at least boot to it's menu?

I hand test every adapter before shipping but there is always a chance of failure. Another test you can do is to try out the `pce_mouse.pce` test rom. This rom will display the raw values the controller/adapter is sending to the console. When no button is pressed, all the values should be zero (0).  If values appear without button presses, then the adapter or cable may be faulty. To rule that out, you can download that `pce_mouse.pce` test rom here: https://github.com/dshadoff/PCE_Mouse_Test/blob/main/src/pce_mouse.pce

p0ojisan, 2024-01-18:
Thank you for feedback.

I own physical copies of 30 games approximately, but this issue happens only when playing Youkai Douchuuki. And also, I can replicate this issue with either physical copies or EverDrive.

My setup is USB2PCE + Turbo EverDrive Pro + CoreGrafx II. As video output, I confirmed this issue with either CVBS+RetroTINK 4K or EDFX + RetroTINK 4K. But I'm not sure if that matters or not.

> Also you mentioned the console does not boot when USB2PCE is connected.

Sorry for confusing. The console **does boot** even with USB2PCE connected. But the screen is black out. I can open EverDrive menu by pressing the physical button on it, but In-Game button combo doesn't work.

Ｗhen I load `pce_mouse.pce`, all the values are zero. But if I try plugging/unplugging, those values are randomly updated.

Here, I recorded video again with screen capturing.
https://www.youtube.com/watch?v=IxPdggBhAEI

RobertDaleSmith, 2024-01-18:
Thanks so much for taking the time to make the video. That really helps me see what is going on, but that is very strange behavior. Everything I saw within the `pce_mouse.pce` rom appeared to be normal behavior though.

This is a really weird one for sure. Initially I would have guesses this was an issue specifically between EverDrive and USB2PCE but it occurs with the original game. I went ahead and ordered a physical copy of the game so I can test further.

p0ojisan, 2024-01-19:
Appreciate your kindly support! Other than this issue, it's awesome product.

I will also buy other titles and verify. Once I get any clue, will update here :)

RobertDaleSmith, 2024-01-25:
Okay I got a physical copy of the game and I am able to replicate the issue locally now.

I will do some troubleshooting to see if I can pinpoint the cause of this issue. May be something different with how this game polls the controller data. So I’ll connect the scope and see.