joypad-ai/joypad-os #38: Turbo Everdrive Pro In-Game Menu Not Working

Outkast92885, 2024-06-23:
The In-Game Menu button combos on the Turbo Everdrive Pro are not working with the 8bitdo PCE controller. I'm on the latest firmware on the USB2PCE and on the Turbo Everdrive Pro.

RobertDaleSmith, 2024-06-23:
Only the default Turbo Everdrive Pro in-game menu combos will work. Also support can vary from one game to another. This is largely due to how the TEP checks for the button combos and its incompatibility with the mutli-tap, which the USB-2-PCE will behave as if the specific game booted polls for it.

I spoke with creator of TEP and sounds like a pretty hacky approach to detecting the button combo that is unreliable when multi-taps are in use. I made a patch to try to patch around the issue but it may not be perfect.

Did it previously work for you and just recently broke? And/or does it not work on some games but it does on others?

Outkast92885, 2024-06-23:
I just recently updated the USB2PCE. I was on the original firmware that was shipped to me. I think I purchased on the first batch. I just tried it again, and it does work on a few games, not many. I appreciate your response on this. Thank you for letting me know.