joypad-ai/joypad-os #39: USB hub support is broken

Tronicgg, 2025-02-05:
I went through the resolved issues and you marked USB hub issues as resolved. I don't think this is resolved. Been testing with 4 different hubs with latest release and there are plenty of issues. Cannot get 2-3 players to work consistently, 4 is impossible. Sometimes it simply does not work but after a couple restarts and disconnecting and connecting controllers, adapter and USB hub I get it to work for a bit (only 3 player, mind you). I would say this is most definitely broken. Right now as it stands I am having extreme difficulties getting 2 players to work consistently with two 8bitdo PCE controllers.

Controllers are 2 8bitdo PCE + 2 8bitdo Arcade sticks.

Tested with following hubs:
USB 2.0 3-ports.
USB 2.0 4-ports.
USB 3.0 4-ports + external power.
USB 3.0 3-ports + external power.

Any ideas? Or particular hub you have tested successfully? I have tested will all hubs I could get hold of.

RobertDaleSmith, 2025-02-05:
I will double check this later this evening to confirm. No recent changes should have impacted hub support but were you able to have any luck with older firmware versions?

I know for sure it was working in the past with the few USB hubs I have.

I use this one most the time: https://www.amazon.com/Splitter-Expansion-Adapter-Compatible-Console/dp/B097SY76JR/ref=asc_df_B097SY76JR

Tronicgg, 2025-02-05:
I was using firmware from late 2023 due to earlier issues with diagonals, but that firmware also had hub support broken. Just recently updated to latest version and retested but same issues tbh. 

Can you test 2-3-4 players and hot swapping and let me know please?

Tronicgg, 2025-02-14:
Any news on this @RobertDaleSmith ?

RobertDaleSmith, 2025-02-14:
So sorry for slow response. I was able to test it out a few days ago. The two USB hubs I tried worked without any issues. The non-powered USB hub did struggle to support multiple PS5 controllers though. I used a really old Targus 4-port USB 2.0 hub and a small USB hub for Xbox Series S consoles. I tested with the latest firmware. I tried 4 players and hot-swapping auto reassigns players as expected.

For quick and easy testing, I am using this controller tester rom called pce_mouse.pce. 
https://github.com/dshadoff/PCE_Mouse_Test/blob/main/src/pce_mouse.pce

If you can load roms on your console, then I would give this a try for quick verification of buttons received.

I'm suspecting the console is struggling to power more than one controller. But you did mention that you tried a powered hub, so trying to think what else it could be.

Tronicgg, 2025-02-14:
@RobertDaleSmith Assuming the test was done with PS5 controllers wired to the hub. Do you have the ability to test with 8bitdo PCE wireless 2.4g controllers? I mean, it could be that I'm the unluckiest man ever here and none of my four USB hubs work, but I tend to think there is something wrong when using these specific 8bitdo controllers and the adapter.

RobertDaleSmith, 2025-02-15:
One of the four controllers I used was the 8bitdo PCE 2.4g controller. 

This controller does have different input modes that are undocumented. I should have support for all modes but may be worth trying different modes to see if its specific to one of them.

https://x.com/RobertDaleSmith/status/1570563632114831360

RobertDaleSmith, 2025-12-28:
@Tronicgg I recently did major overhaul of the firmware and updated to latest TinyUSB library which may or may not have improved USB hub support in general. I'm curious if this is still an issue on your end.

Tronicgg, 2025-12-28:
> [@Tronicgg](https://github.com/Tronicgg) I recently did major overhaul of the firmware and updated to latest TinyUSB library which may or may not have improved USB hub support in general. I'm curious if this is still an issue on your end.

@RobertDaleSmith Yep, still an issue on my end. I went and bought the hub you recommended ([https://www.amazon.com/Splitter-Expansion-Adapter-Compatible-Console/dp/B097SY76JR]) but same issues. I will update, test and report when I get a chance.