joypad-ai/joypad-os #40: Linking fails due to multiple definitions of global variables

timonsku, 2025-03-21:
Various globally declared variables such as `is_fun` are defined multiple times in various headers in the common include and the linker fails.
I'm not quite sure how it compiled in the past as is as this seems to have been the case for a long time. Was this relying on default behavior of an older toolchain?

RobertDaleSmith, 2025-03-22:
I have had others report this but I don't seem to get this error when building in my environment or the docker build container. I'm not certain what the cause is but suspecting differing gcc versions. I'm running `gcc version 10.2.1 20210110 (Debian 10.2.1-6)` I would like to get this ironed out though if you figure it out.

timonsku, 2025-03-22:
Ooh ok, then the cause will be the behavior change in GCC 11. They made -fno-common the default.
I tried with GCC 12 and 14, which uses that new default. It indeed compiles with -fcommon set.
 Here some details: https://trofi.github.io/posts/245-fno-common-legacy.html

I tried to wrap my head around the error but I can't quite make out the root cause. You correctly defined the affected variables as extern in all relevant places from what I can see. The issue seems to be some sort of circular inclusion issue as it pops up in pretty much every compilation unit.